Twisted-Fields/rp2040-motor-controller #1: Noob question gerber files

edmundasa, 2022-06-11:
Hi, I found your design interesting and would like to try it out, but Im not able to order boards as I cant seem to find how to order. Im missing gerber files zip jlcpcb requires.
Could you add this file for users like me who havent used to pcb design?

Thank you for you help.
Edmundas

tlalexander, 2022-06-11:
Thank you Edmundas! The design is nearly complete, but not 100% done yet. I will leave this issue open, and when it is ready to order I will include those files and let you know.

tlalexander, 2022-06-12:
I've actually opened a new issue #2 to serve as a notification for when gerbers are ready in the git repo, and when we have tested boards for functionality and have some to distribute to others. So I will close this issue, but please subscribe to issue #2 to be notified of new updates. Thanks!

closed 2022-06-12 by tlalexander