Twisted-Fields/rp2040-motor-controller #2: Subscribe to this issue to be notified of board availability.

tlalexander, 2022-06-12:
When the layout is completed and boards are available, I will update this issue. I will update once when we have finalized things and ordered the first batch of boards to test, and subsequently as we make progress on verification and order more boards to be distributed to community members to help with firmware development. We may sell boards in the future, or encourage someone else in the community to take on that work, after we have verified the design.

tlalexander, 2022-07-04:
I have ordered ten boards through JLCPCB assembly service, which should arrive in about two weeks. I would like to send free boards to people who have the skills and availability to work on firmware porting. Please comment here if you would like to get one.
We will be working with this code base:
https://docs.simplefoc.com/rpi_mcu

ScottMonaghan, 2022-07-05:
I'd love to help with this. I'll contact you via DM on Twitter.

caswal, 2022-07-20:
Hi,

Would love to help with porting SimpleFOC Firmware to the board. I'm helping a friend with his EV Hillclimb car (https://www.youtube.com/watch?v=rItlZ7KyBSk), so was wanting a project based on SimpleFOC to get more of an understanding of how FOC works.

tlalexander, 2022-07-20:
Great! Much appreciated. This should be enough testers for the initial run (I have to build and ship some test stands for them) so I will put a hold on any future help but anyone who expressed interest at this point should be able to get one. Also on the SimpleFOC forum I got the person who originally made the RP2040 port onboard with helping. So we should knock this port out of the park!

I just received 10 assembled boards. I will be testing them soon. I sure hope they work! 😂

tlalexander, 2022-09-17:
Update:
We have completed initial firmware bring up and confirmed the boards are working. Hooraaay! I've just finished soldering phase and power leads on the rest of the boards we have. I've built one more bench test stand today that is ready to go and I should be able to build the other some time next week. Please email me at the gmail address listed on my github profile (the tlalexander one at gmail) with your shipping address, and do let me know if your availability to look at this has changed at all. The basic port is working great but there is plenty more work to be done. Thanks so much for your help!

ScottMonaghan, 2022-09-17:
Hey Taylor,

I’m definitely in.

I’m just coming up from a week of covid.

I’m going to put a reminder to get back to you next week.

eric-schleicher, 2023-05-10:
following up from twitter to help evaluate if you'll let me.  one would go to steering wheel/servo;  the other to large form factor 3d scanner.