Twisted-Fields/rp2040-motor-controller #3: BOM upload

samuk, 2022-06-14:
I was curious about getting some boards made up by JLPCB.

I opened the project in Kicad and used this plugin https://github.com/Bouni/kicad-jlcpcb-tools to generate Gerbers & BOM. eg [BOM-RP2040_motor.csv](https://github.com/Twisted-Fields/rp2040-motor-controller/files/8898464/BOM-RP2040_motor.csv)

When I upload these files to JLPCB 56 of the components are not automatically detected.   
[image]
I guess this plugin may be buggy/incomplete? Is there a better way to do this?

Could you upload a BOM where JLPCB auto-detects all the components?

nathmo, 2022-06-15:
Something else related to the BOM : would it be possible to mention the approximate total of the BOM on the README.md ? How much does this controller cost ? 10 ? 20 ? 50 ? 100? 200?

samuk, 2022-06-15:
Posted

closed 2022-06-15 by samuk