Twisted-Fields/rp2040-motor-controller #5: Improvement proposition : mounting hole

nathmo, 2022-06-19:
This is just an idea : what about 4 mounting holes on the board so that it can be cleanly attached in whatever enclosure it might be used  ?

james-yoo, 2022-06-19:
I agree. Nice idea!

tlalexander, 2022-06-21:
Yes I have probably obsessed a bit too much about it being the exact size as a credit card. I had planned to use a 3D printed frame to hold it, but many people have asked about mounting holes. I will see about fitting them in.

nathmo, 2022-06-21:
Good luck re routing everything. But again, really nice works. I can't wait to find a bit of time to make my own version of it (single channel that mount on the back of the brushless with a hall sensor for position sensing) (is the PDF schematic still up to date ?) (I think there is a plugin that regenerate the gerber and pdf everything there is a push / merge to the branch)

tlalexander, 2022-07-04:
Thank you! I believe the PDF schematic is up to date now. But if you know the name of the auto update plugin that would be appreciated!

nathmo, 2022-07-05:
https://github.com/marketplace/actions/kicad-exports