Twisted-Fields/rp2040-motor-controller #6: Improvement idea: CAN bus

vortex505, 2022-07-02:
Great board!  CAN bus is commonly used with industrial motor drives and is robust to noise inevitably produces in robotic systems. With a dual core MCU, you might consider running CANopen device code on one core while the other is tasked with controlling the motor.

tlalexander, 2022-07-04:
Thank you! Yes we will make a version with CAN bus. A daughter board would also be interesting. I wanted to keep our initial version simple (and compact) and go from there, and it seems CAN raises the cost a bit. But we will make a version specific to our farm robot with CAN and some other changes which could be back ported to a generic board.
I am not familiar with CANopen but I will take a look!

runger1101001, 2022-09-18:
There's also a couple of CAN implementations for SimpleFOC... unfortunately the RP2040 doesn't have a CAN peripheral? But it can do it via PIO, it seems. Other option is to use an SPI based controller, but those are more expensive than simple transceivers, and would require another driver layer.